jdlangs/RobotOS.jl #7: Cannot construct message of type std_msgs.msg.Float64

ashwincarvalho, 2015-05-07:
I am trying to write a publisher node in Julia which broadcasts a message of type Float64. However, the constructor method for std_msgs.msg.Float64 doesn't seem to work. Here's the code that produces the error:
```
using RobotOS
@rosimport std_msgs.msg: Float64
gentypes()
using std_msgs.msg
pub_msg = std_msgs.msg.Float64()
```
Output:
```
ERROR: `typezero` has no method matching typezero(::Type{Float64})
 in Float64MultiArray at no file
```

This is the case with any of the '_ros_builtin_types_' defined in gentypes.jl. For example, if I run the above code with Float64MultiArray instead of Float64, I do not get this error.

@tkelman pointed out that the issue is caused due to std_msgs.msg.Float64 not satisfying the qualifier on Line 25 of gentypes.jl:
```
typezero{T<:Real}(::Type{T}) = zero(T)
```
Replacing this line with:
```
typezero{T}(::Type{T}) = zero(T)
```
gives me the following error with using the method zero():
```
ERROR: `zero` has no method matching zero(::Type{Float64})
 in Float64 at no file
```

phobon, 2015-05-07:
I'm not totally sure what it's doing because `typezero` is only supposed to be invoked for the primitive types that make up the message (`float64 data` in this case). I'm guessing when `std_msgs` is generated, the `Float64` ROS message type wipes out the Julia type. If that's the case, it's quite unfortunate that Julia and ROS used the exact same names and I'm not quite sure what the best workaround should be.

phobon, 2015-05-07:
Probably I'll have to break API compatibility and generate the primitive message types as `ROSFloat64`, etc...

mcorah, 2015-05-07:
If the call is made like `Base.Float64` you should be able to resolve that ambiguity.

phobon, 2015-05-08:
Yes, that does fix it, though you need to carefully insert `Base.` in a few other places too.

But I think it's probably too potentially dangerous/confusing to be overwriting such a core element of Base. I would also expect the numeric primitive wrapper messages from `std_msgs` to be very rarely used.

@ashwincarvalho, what would be your preference for your use case? Keeping the default name or keeping easy access to the Julia type?

mcorah, 2015-05-08:
I think see what you mean. The message types are exported which could break code in the importing workspace later on. The user however has the option of loading the message with either `using` or `import`. It would probably be sufficient to clearly document that naming conflicts will occur and to use `import` rather than `using` in that case. Alternatively, not exporting the message type would ensure that the user is not accidentally overwriting core datatypes in their namespace.

ashwincarvalho, 2015-05-08:
@phobon: The main reason we're interested in the ROS-Julia integration is to be able to use the Julia optimization packages. I do not expect to use the ROS primitive message types too much. It'll be nice to have that basic functionality though. Keeping easy access to the builtin Julia types and renaming the ROS message types seems like the better option to me.

phobon, 2015-05-08:
@mcorah, there's also just the programmer cost of having to think which version of `Float64` is being used. And just due to the message nature, you are guaranteed to have to use both in your code (though you might not have to type out Julia's `Float64`).

It's a pretty minor corner of ROS so either way probably wouldn't cause too much grief, but I think I'd also choose the rename.

phobon, 2015-05-08:
What would you two prefer:

- `ROSFloat64`, `ROSInt8`, etc...
- `Float64Msg`, `Int8Msg`, ...
- something else?

I'm leaning towards the `*Msg` names right now.

ashwincarvalho, 2015-05-08:
Yes, that would be my preference too. Thanks.

commit 5de5e096 (2015-05-14): Change generation of messages that conflict with built-ins

Closes #7
closed 2015-05-14 by phobon