jdlangs/RobotOS.jl #9: `ERROR bad callback: <PyCall.jlwrap jl_cb>` when trying to use with `sensor_msgs/Image`

nstiurca, 2015-11-12:
I tried modifying the example code to use `sensor_msgs/Image`, but I am getting errors when trying to run, and the callback isn't called. Any thoughts?

------
Source code:

    #!/usr/bin/env julia

    using RobotOS
    @rosimport sensor_msgs.msg.Image
    rostypegen()
    using sensor_msgs.msg

    function callback(img::Image, pub_obj::Publisher{Image})
        println(img.header)
        publish(pub_obj, img)
    end

    function loop(pub_obj)
        loop_rate = Rate(5.0)
        while ! is_shutdown()
    #         npt = Point(rand(), rand(), 0.0)
    #         publish(pub_obj, npt)
            rossleep(loop_rate)
        end
    end

    function main()
        init_node("inpaint_depth")
        pub = Publisher{Image}("depth_inpainted", queue_size=10)
        sub = Subscriber{Image}("depth", callback, (pub,), queue_size=10)
        loop(pub)
    end

    if ! isinteractive()
        main()
    end

-----
Error:

    $ ./inpaint_depth.jl depth:=/camera/depth_registered/image_raw
    [ERROR] [WallTime: 1447366470.036321] bad callback: <PyCall.jlwrap jl_cb>
    Traceback (most recent call last):
      File "/opt/ros/indigo/lib/python2.7/dist-packages/rospy/topics.py", line 711, in _invoke_callback
        cb(msg)
    TypeError: an integer is required

    [ERROR] [WallTime: 1447366470.307730] bad callback: <PyCall.jlwrap jl_cb>
    Traceback (most recent call last):
      File "/opt/ros/indigo/lib/python2.7/dist-packages/rospy/topics.py", line 711, in _invoke_callback
        cb(msg)
    TypeError: an integer is required

    [ERROR] [WallTime: 1447366470.869406] bad callback: <PyCall.jlwrap jl_cb>
    Traceback (most recent call last):
      File "/opt/ros/indigo/lib/python2.7/dist-packages/rospy/topics.py", line 711, in _invoke_callback
        cb(msg)
    TypeError: an integer is required

    < repeats with every incoming Image >

nstiurca, 2015-11-12:
In case it's relevant, I'm using Julia v0.4, Python v2.7, and ROS Indigo.

phobon, 2015-11-12:
Thanks for the report. Could you make a small bag file of the image messages that are failing so I can be sure I'm exactly reproducing the problem?

nstiurca, 2015-11-13:
https://www.assembla.com/spaces/cis700-snap/documents/be0R-MIAGr5zGHdmr6CpXy/download/be0R-MIAGr5zGHdmr6CpXy

nstiurca, 2015-11-13:
<pre><code>
$ rosbag info /tmp/2015-11-12-20-43-40.bag 
path:         /tmp/2015-11-12-20-43-40.bag
version:      2.0
duration:     5.3s
start:        Nov 12 2015 20:43:41.27 (1447379021.27)
end:          Nov 12 2015 20:43:46.57 (1447379026.57)
size:         1.9 MB
messages:     55
compression:  bz2 [28/28 chunks; 5.98%]
uncompressed: 32.2 MB @   6.1 MB/s
compressed:    1.9 MB @ 372.1 KB/s (5.98%)
types:        sensor_msgs/Image [060021388200f6f0f447d0fcd9c64743]
topics:       /camera/depth_registered/image_raw   55 msgs    : sensor_msgs/Image
</code></pre>

phobon, 2015-11-13:
Thanks, I'll see if I have some time to try a fix by this weekend.

phobon, 2015-11-13:
It turned out to be an upstream issue with `PyCall`. Specifically, it didn't know how to transparently convert a python string (rospy's byte storage method) to Julia's Vector{UInt8}. That is fixed now. Try running `Pkg.checkout(PyCall)` and see if it works.

I'll also add some `Image` tests since apparently I didn't do a thorough enough job in manual testing before.

nstiurca, 2015-11-14:
Much better, but still a strange error:

    $ julia inpaint_depth.jl depth:=/camera/depth_registered/image_raw
    std_msgs.msg[ERROR] [WallTime: 1447462026.962798] bad callback: <PyCall.jlwrap jl_cb>
    Traceback (most recent call last):
      File "/opt/ros/indigo/lib/python2.7/dist-packages/rospy/topics.py", line 711, in _invoke_callback
        cb(msg)
    MemoryError: Julia exception: OutOfMemoryError()

    std_msgs.msg.Header(0x0000002a,RobotOS.Time(1447379021,246340814),"camera_rgb_optical_frame")
    std_msgs.msg.Header(0x0000002c,RobotOS.Time(1447379021,385057292),"camera_rgb_optical_frame")
    std_msgs.msg.Header(0x0000002e,RobotOS.Time(1447379021,448288826),"camera_rgb_optical_frame")
    std_msgs.msg.Header(0x00000030,RobotOS.Time(1447379021,613769898),"camera_rgb_optical_frame")
    std_msgs.msg.Header(0x00000031,RobotOS.Time(1447379021,648030688),"camera_rgb_optical_frame")
    std_msgs.msg.Header(0x00000032,RobotOS.Time(1447379021,748913319),"camera_rgb_optical_frame")
    std_msgs.msg.Header(0x00000033,RobotOS.Time(1447379021,848617210),"camera_rgb_optical_frame")
    std_msgs.msg.Header(0x00000035,RobotOS.Time(1447379021,984761540),"camera_rgb_optical_frame")

nstiurca, 2015-11-14:
Ie, the first callback fails, but the rest are fine. Race condition?

phobon, 2015-11-16:
I have not ever seen an `OutOfMemoryError` before. Is it always repeatable? Are you running it on a somewhat resource-limited system?

nstiurca, 2015-11-16:
Always repeatable on a sturdy gaming laptop. Is it not reproducible for
you? There might be some other package on my system needing updating to
master branch or such.

For the record,

    julia> Pkg.status()
    11 required packages:
     - CUDA                          0.1.0
     - Color                         0.2.11
     - Debug                         0.1.6
     - FactCheck                     0.4.1
     - IJulia                        1.1.8
     - ImageView                     0.1.18
     - Images                        0.5.0
     - Jewel                         1.0.7
     - Mocha                         0.1.0+             opencl
     - Quaternions                   0.0.4
     - RobotOS                       0.3.5
    42 additional packages:
     - BinDeps                       0.3.19
     - Blosc                         0.1.4
     - CLBLAS                        0.1.0+
deprecated-release-fix
     - Cairo                         0.2.31
     - Calculus                      0.1.14
     - ColorTypes                    0.2.0
     - ColorVectorSpace              0.1.0
     - Colors                        0.6.0
     - Compat                        0.7.7
     - Compose                       0.3.18
     - Conda                         0.1.8
     - DataStructures                0.3.13
     - Dates                         0.4.4
     - Docile                        0.5.19
     - DualNumbers                   0.1.5
     - FileIO                        0.0.3
     - FixedPointNumbers             0.1.1
     - Graphics                      0.1.3
     - HDF5                          0.5.6
     - IniFile                       0.2.4
     - Iterators                     0.1.9
     - JLD                           0.5.6
     - JSON                          0.5.0
     - JuliaParser                   0.6.3
     - LNR                           0.0.2
     - Lazy                          0.10.1
     - Logging                       0.2.0
     - MacroTools                    0.2.0
     - NaNMath                       0.1.1
     - Nettle                        0.2.0
     - OpenCL                        0.3.6+             master
     - PyCall                        1.2.0+             master
     - Reexport                      0.0.3
     - Requires                      0.2.1
     - SHA                           0.1.2
     - SIUnits                       0.0.6
     - TexExtensions                 0.0.3
     - Tk                            0.3.6
     - URIParser                     0.1.1
     - Winston                       0.11.13
     - ZMQ                           0.3.0
     - Zlib                          0.1.12

    julia> versioninfo()
    Julia Version 0.4.0
    Commit 0ff703b* (2015-10-08 06:20 UTC)
    Platform Info:
      System: Linux (x86_64-linux-gnu)
      CPU: Intel(R) Core(TM) i7-4710HQ CPU @ 2.50GHz
      WORD_SIZE: 64
      BLAS: libopenblas (NO_LAPACK NO_LAPACKE DYNAMIC_ARCH NO_AFFINITY
Haswell)
      LAPACK: liblapack.so.3
      LIBM: libopenlibm
      LLVM: libLLVM-3.3

-- Nicu Știurcă
\\//./ Live long and prosper.

nstiurca, 2015-11-16:
Always repeatable on a sturdy gaming laptop. Is it not reproducible for you? There might be some other package on my system needing updating to master branch or such.

For the record,

    julia> versioninfo()
    Julia Version 0.4.0
    Commit 0ff703b* (2015-10-08 06:20 UTC)
    Platform Info:
      System: Linux (x86_64-linux-gnu)
      CPU: Intel(R) Core(TM) i7-4710HQ CPU @ 2.50GHz
      WORD_SIZE: 64
      BLAS: libopenblas (NO_LAPACK NO_LAPACKE DYNAMIC_ARCH NO_AFFINITY Haswell)
      LAPACK: liblapack.so.3
      LIBM: libopenlibm
      LLVM: libLLVM-3.3

    julia> Pkg.status()
    11 required packages:
     - CUDA                          0.1.0
     - Color                         0.2.11
     - Debug                         0.1.6
     - FactCheck                     0.4.1
     - IJulia                        1.1.8
     - ImageView                     0.1.18
     - Images                        0.5.0
     - Jewel                         1.0.7
     - Mocha                         0.1.0+             opencl
     - Quaternions                   0.0.4
     - RobotOS                       0.3.5
    42 additional packages:
     - BinDeps                       0.3.19
     - Blosc                         0.1.4
     - CLBLAS                        0.1.0+             deprecated-release-fix
     - Cairo                         0.2.31
     - Calculus                      0.1.14
     - ColorTypes                    0.2.0
     - ColorVectorSpace              0.1.0
     - Colors                        0.6.0
     - Compat                        0.7.7
     - Compose                       0.3.18
     - Conda                         0.1.8
     - DataStructures                0.3.13
     - Dates                         0.4.4
     - Docile                        0.5.19
     - DualNumbers                   0.1.5
     - FileIO                        0.0.3
     - FixedPointNumbers             0.1.1
     - Graphics                      0.1.3
     - HDF5                          0.5.6
     - IniFile                       0.2.4
     - Iterators                     0.1.9
     - JLD                           0.5.6
     - JSON                          0.5.0
     - JuliaParser                   0.6.3
     - LNR                           0.0.2
     - Lazy                          0.10.1
     - Logging                       0.2.0
     - MacroTools                    0.2.0
     - NaNMath                       0.1.1
     - Nettle                        0.2.0
     - OpenCL                        0.3.6+             master
     - PyCall                        1.2.0+             master
     - Reexport                      0.0.3
     - Requires                      0.2.1
     - SHA                           0.1.2
     - SIUnits                       0.0.6
     - TexExtensions                 0.0.3
     - Tk                            0.3.6
     - URIParser                     0.1.1
     - Winston                       0.11.13
     - ZMQ                           0.3.0
     - Zlib                          0.1.12

phobon, 2015-11-16:
No, all the messages in the bag file are received fine by the callback on my end and I tried it on several different julia versions (0.4.0, 0.4.1, 0.5). There's probably a bug somewhere but it seems to be platform-specific. If you can reproduce it on another machine that would be good to know.

nstiurca, 2015-11-16:
Ok, good to know. Since it seems to be specific to my setup, I will try to
look into this tonight.

-- Nicu Știurcă
\\//./ Live long and prosper.