jdlangs/RobotOS.jl #10: Error in Pkg.test("RobotOS") and "Float64Msg" not found

shionlau, 2016-02-23:
I installed RobotOS on Julia and ran Pkg.test("RobotOS"), which gave me the following error:

> INFO: Testing RobotOS
> [INFO] [WallTime: 1456200498.459841] testing: info
> [WARN] [WallTime: 1456200498.464155] testing: warn
> [ERROR] [WallTime: 1456200498.467667] testing: err
> [FATAL] [WallTime: 1456200498.471353] testing: fatal
> Creating new package: geometry_msgs.msg
> Message type import: geometry_msgs.msg.PoseStamped
> Importing python package: geometry_msgs.msg
> Dependency: std_msgs/Header
> 	Creating new package: std_msgs
> 	Message type import: std_msgs.msg.Header
> 	Importing python package: std_msgs.msg
> Dependency: geometry_msgs/Pose
> 	Message type import: geometry_msgs.msg.Pose
> 	Dependency: geometry_msgs/Point
> 		Message type import: geometry_msgs.msg.Point
> 	Dependency: geometry_msgs/Quaternion
> 		Message type import: geometry_msgs.msg.Quaternion
> Message type import: geometry_msgs.msg.Vector3
> Creating new package: std_srvs.srv
> Service type import: std_srvs.srv.Empty
> Importing python package: std_srvs.srv
> Creating new package: nav_msgs.srv
> Service type import: nav_msgs.srv.GetPlan
> Importing python package: nav_msgs.srv
> Dependency: geometry_msgs/PoseStamped
> Dependency: geometry_msgs/PoseStamped
> Dependency: nav_msgs/Path
> 	Message type import: nav_msgs.msg.Path
> 	Importing python package: nav_msgs.msg
> 	Dependency: std_msgs/Header
> 	Dependency: geometry_msgs/PoseStamped[]
> Message type import: std_msgs.msg.Empty
> Message type import: std_msgs.msg.Float64
> Message type import: std_msgs.msg.String
> Creating new package: fake_msgs.msg
> Message type import: fake_msgs.msg.FakeMsg
> Importing python package: fake_msgs.msg
> PyError (:PyImport_ImportModule) <type 'exceptions.ImportError'>
> ImportError('No module named fake_msgs.msg',)
> Message type import: std_msgs.msg.FakeMsg
> Service type import: nav_msgs.srv.GetPlanRequest
> Building package: std_srvs
> 	submodule: std_srvs.srv
> 		Type: EmptyRequest
> 		Type: EmptyResponse
> Building package: std_msgs
> 	submodule: std_msgs.msg
> 		Type: Header
> 			seq :: uint32
> 			stamp :: time
> 			frame_id :: string
> 		Type: Empty
> 		Type: Float64
> 			data :: float64
> 		Type: String
> 			data :: string
> Building package: geometry_msgs
> 	submodule: geometry_msgs.msg
> 		Type: Point
> 			x :: float64
> 			y :: float64
> 			z :: float64
> 		Type: Quaternion
> 			x :: float64
> 			y :: float64
> 			z :: float64
> 			w :: float64
> 		Type: Pose
> 			position :: geometry_msgs/Point
> 			orientation :: geometry_msgs/Quaternion
> 		Type: PoseStamped
> 			header :: std_msgs/Header
> 			pose :: geometry_msgs/Pose
> 		Type: Vector3
> 			x :: float64
> 			y :: float64
> 			z :: float64
> Building package: nav_msgs
> 	submodule: nav_msgs.msg
> 		Type: Path
> 			header :: std_msgs/Header
> 			poses :: geometry_msgs/PoseStamped[]
> 	submodule: nav_msgs.srv
> 		Type: GetPlanRequest
> 			start :: geometry_msgs/PoseStamped
> 			goal :: geometry_msgs/PoseStamped
> 			tolerance :: float32
> 		Type: GetPlanResponse
> 			plan :: nav_msgs/Path
> Building package: fake_msgs
> Creating <std_msgs.msg.Float64Msg> publisher on topic: 'x'
> Creating <std_msgs.msg.Float64Msg> subscriber on topic: 'x'
> Message received, time: -1456200504277520895 nanoseconds
> Creating <geometry_msgs.msg.Vector3> publisher on topic: 'vectors'
> Creating <geometry_msgs.msg.PoseStamped> subscriber on topic: 'poses'
> Received 0 / 10
> ERROR: LoadError: LoadError: test failed: 0 == 10
>  in expression: length(msgs) == Nmsgs
>  in error at ./error.jl:21
>  in default_handler at test.jl:30
>  in do_test at test.jl:53
>  in include at ./boot.jl:261
>  in include_from_node1 at ./loading.jl:304
>  in include at ./boot.jl:261
>  in include_from_node1 at ./loading.jl:304
>  in process_options at ./client.jl:280
>  in _start at ./client.jl:378
> while loading /home/vdl/.julia/v0.4/RobotOS/test/pubsub.jl, in expression starting on line 42
> while loading /home/vdl/.julia/v0.4/RobotOS/test/runtests.jl, in expression starting on line 11
> ===============================[ ERROR: RobotOS ]===============================
> 
> failed process: Process(`/usr/bin/julia --check-bounds=yes --code-coverage=none --color=yes /home/vdl/.julia/v0.4/RobotOS/test/runtests.jl`, ProcessExited(1)) [1]
> 
> ================================================================================
> ERROR: RobotOS had test errors
>  in error at ./error.jl:21
>  in test at pkg/entry.jl:803
>  in anonymous at pkg/dir.jl:31
>  in cd at file.jl:22
>  in cd at pkg/dir.jl:31
>  in test at pkg.jl:71

It fails on data transmission.

Then I ran a Julia script that was used for publishing to ROS. However, the following error occurred:

> ERROR: LoadError: KeyError: 'Float64Msg' in package 'std_msgs.msg' not found
 in _pyvars at /home/vdl/.julia/v0.4/RobotOS/src/gentypes.jl:199
 in addtype! at /home/vdl/.julia/v0.4/RobotOS/src/gentypes.jl:155
 in _usepkg at /home/vdl/.julia/v0.4/RobotOS/src/gentypes.jl:121
 in include at ./boot.jl:261
 in include_from_node1 at ./loading.jl:304
while loading /home/vdl/catkin_ws/src/mpc_julia/scripts/controller.jl, in expression starting on line 4

I don't quite understand this error since 'Float64Msg' type publisher and subscriber have passed the test before. What could be wrong here?

System:
indigo 1.11.16
Ubuntu 14.04

installed packages:

> 6 required packages:
 - Interpolations                0.3.4
 - Ipopt                         0.2.1
 - JuMP                          0.11.3
 - MAT                           0.2.14
 - Polynomials                   0.0.5
 - RobotOS                       0.3.5
22 additional packages:
 - ArrayViews                    0.6.4
 - AxisAlgorithms                0.1.4
 - BinDeps                       0.3.21
 - Blosc                         0.1.4
 - Calculus                      0.1.14
 - Compat                        0.7.11
 - Conda                         0.1.9
 - DataStructures                0.4.3
 - Dates                         0.4.4
 - Docile                        0.5.23
 - DualNumbers                   0.2.1
 - HDF5                          0.5.8
 - JSON                          0.5.0
 - MathProgBase                  0.4.2
 - NaNMath                       0.2.0
 - PyCall                        1.4.0
 - Ratios                        0.0.4
 - ReverseDiffSparse             0.3.0
 - SHA                           0.1.2
 - URIParser                     0.1.2
 - WoodburyMatrices              0.1.5
 - Zlib                          0.1.12

phobon, 2016-02-23:
Hello @shionlau, 

For your first question, the package is not set up to run `Pkg.test` right after install because it uses another ROS node to test communication. If you run the `test/echonode.py` file before, it should pass tests.

For your second question, did you by chance put `@rosimport std_msgs.msg.Float64Msg` instead of `Float64` in your script? `@rosimport` calls refer to the proper message names and only the Julia types are potentially renamed.

shionlau, 2016-02-23:
Hi @phobon,

Thanks for your quick reply! After running test/echonode.py, it passes the test.

I was using @rosimport std_msgs.msg.Float64Msg before but got the error. After changing to @rosimport std_msgs.msg.Float64 it works now.

closed 2016-02-23 by shionlau

phobon, 2016-02-23:
Glad it could be resolved quickly!

ga72kud, 2022-04-27:
Have similar problem. I think it worked for me too, but still I am wondering about [ERROR] [1651077579.813758]: testing: err
[FATAL] [1651077579.820518]: testing: fatal

julia> Pkg.test("RobotOS")
     Testing RobotOS
      Status `/tmp/jl_SHCjEH/Project.toml`
  [438e738f] PyCall v1.93.1
  [22415677] RobotOS v0.7.2
  [8dfed614] Test `@stdlib/Test`
      Status `/tmp/jl_SHCjEH/Manifest.toml`
  [8f4d0f93] Conda v1.7.0
  [682c06a0] JSON v0.21.3
  [1914dd2f] MacroTools v0.5.9
  [69de0a69] Parsers v2.2.3
  [438e738f] PyCall v1.93.1
  [22415677] RobotOS v0.7.2
  [81def892] VersionParsing v1.3.0
  [0dad84c5] ArgTools `@stdlib/ArgTools`
  [56f22d72] Artifacts `@stdlib/Artifacts`
  [2a0f44e3] Base64 `@stdlib/Base64`
  [ade2ca70] Dates `@stdlib/Dates`
  [f43a241f] Downloads `@stdlib/Downloads`
  [b77e0a4c] InteractiveUtils `@stdlib/InteractiveUtils`
  [b27032c2] LibCURL `@stdlib/LibCURL`
  [8f399da3] Libdl `@stdlib/Libdl`
  [37e2e46d] LinearAlgebra `@stdlib/LinearAlgebra`
  [56ddb016] Logging `@stdlib/Logging`
  [d6f4376e] Markdown `@stdlib/Markdown`
  [a63ad114] Mmap `@stdlib/Mmap`
  [ca575930] NetworkOptions `@stdlib/NetworkOptions`
  [de0858da] Printf `@stdlib/Printf`
  [9a3f8284] Random `@stdlib/Random`
  [9e88b42a] Serialization `@stdlib/Serialization`
  [8dfed614] Test `@stdlib/Test`
  [4ec0a83e] Unicode `@stdlib/Unicode`
  [deac9b47] LibCURL_jll `@stdlib/LibCURL_jll`
  [29816b5a] LibSSH2_jll `@stdlib/LibSSH2_jll`
  [c8ffd9c3] MbedTLS_jll `@stdlib/MbedTLS_jll`
  [14a3606d] MozillaCACerts_jll `@stdlib/MozillaCACerts_jll`
  [83775a58] Zlib_jll `@stdlib/Zlib_jll`
  [8e850ede] nghttp2_jll `@stdlib/nghttp2_jll`
     Testing Running tests...
[INFO] [1651077579.801355]: testing: info
[WARN] [1651077579.807259]: testing: warn
[ERROR] [1651077579.813758]: testing: err
[FATAL] [1651077579.820518]: testing: fatal
Importing new package: geometry_msgs.msg
Message type import: geometry_msgs.msg.PoseStamped
Importing python package: geometry_msgs.msg
Dependency: std_msgs/Header
    Creating new package: std_msgs
    Message type import: std_msgs.msg.Header
    Importing python package: std_msgs.msg
Dependency: geometry_msgs/Pose
    Message type import: geometry_msgs.msg.Pose
    Dependency: geometry_msgs/Point
        Message type import: geometry_msgs.msg.Point
    Dependency: geometry_msgs/Quaternion
        Message type import: geometry_msgs.msg.Quaternion
Message type import: geometry_msgs.msg.Vector3
Importing new package: visualization_msgs.msg
Message type import: visualization_msgs.msg.Marker
Importing python package: visualization_msgs.msg
Dependency: std_msgs/Header
Dependency: geometry_msgs/Pose
Dependency: geometry_msgs/Vector3
Dependency: std_msgs/ColorRGBA
    Message type import: std_msgs.msg.ColorRGBA
Dependency: geometry_msgs/Point[]
Dependency: std_msgs/ColorRGBA[]
[... 235 lines cut ...]
Message received, time: 4795946121 nanoseconds
Got message #9
Message received, time: 4845345020 nanoseconds
Got message #10
Received 10 / 10
Creating <SetBool> service proxy for 'callme'
Waiting for 'callme' service...
Providing <GetPlan> service at 'getplan'
Calling service...
Waiting for service call from echo..
Spinning up callback loop...
GetPlan call received
Message received, time: 9267719030 nanoseconds
Got message #11
Message received, time: 9267778158 nanoseconds
Got message #12
Message received, time: 9267786979 nanoseconds
Got message #13
Message received, time: 9267795086 nanoseconds
Got message #14
Message received, time: 9267802000 nanoseconds
Got message #15
Response sent
Providing <SetBool> service at 'fake'
     Testing RobotOS tests passed