jdlangs/RobotOS.jl #11: Message enums

schmrlng, 2016-03-14:
This is an awesome package - thanks for putting it together!

I skimmed gentypes.jl and it seems that the enums in message definitions (see, e.g., http://docs.ros.org/api/visualization_msgs/html/msg/Marker.html, where in python you might write ```visualization_msgs.msg.Marker.CUBE```, or in C++, ```visualization_msgs::Marker::CUBE```) aren't accessible through the generated Julia type. Before I hardcode anything or define my own julia constants, can you confirm that this is the case?

phobon, 2016-03-16:
Hi @schmrlng, you are correct that I didn't do anything auto-generated for enums. I'm not sure if there would be a natural syntax for it either since Julia types can't have static/class members. Maybe a new function `ros_enum(visualization_msgs.msg.Marker, :CUBE)`? Would be interested to hear your thoughts.

In the meantime, you can access the cached `PyObject` after the type has been generated:

```
py_marker = RobotOS._rospy_objects["visualization_msgs/Marker"]
py_marker[:CUBE]
```

I don't have access to a ROS machine for another week so I can't verify this works. Something along those lines should definitely do the trick though.

schmrlng, 2016-03-16:
Cool, I can confirm that ```RobotOS._rospy_objects["visualization_msgs/Marker"][:CUBE]``` works and returns ```1``` as expected.

I think a solution along the lines of ```ros_enum(visualization_msgs.msg.Marker, :CUBE)``` would be in keeping with Julia's multiple-dispatch philosophy towards member variables/methods. It might also be nice to define a convenience method like:
```julia
import Base.getindex
getindex{T<:RobotOS.MsgT}(::Type{T}, s::Symbol) = ros_enum(T, s)
```
unless you have other future plans for ```getindex``` for ROS types.

phobon, 2016-03-23:
Using `getindex` will be a good solution for now, I think. If Julia goes forward with overloading the dot operator, this will be an obvious usage for it and in the meantime it seems several people use `getindex` with `Symbol`s as the current closest approximation.

The only other idea I can come up with is to parse through the message text stored in the python object and auto-generate some `const` variables in some global scope. There would have to be some renaming or nested module scheme though again due to difference in Julia namespaces. So probably no benefit to doing it that way.

jdlangs, 2016-08-29:
Marking this closed by #12.

closed 2016-08-29 by jdlangs