jdlangs/RobotOS.jl #14: julia node not publishing to topic when initiated from roslaunch file on ARM

jgonzal13, 2016-04-30:
great package, thanks for putting it together!

i have a julia node running optimization code on an ARM machine; the node publishes the output on a topic using a custom message type.
when i launch the julia node from the command line using 
`rosrun <package> <file_name>.jl` 
the julia node publishes to the specified topic.
however, when the julia node is included inside a launch file, the julia node appears not publish any messages.

do you have any thoughts on what may be causing this behavior?

phobon, 2016-05-02:
That's pretty bizarre. Are you not able to make the same thing happen with a similar python ROS node? Also, does it depend at all on the optimization in the Julia code? I.e., what's the simplest Julia node where this happens?

jgonzal13, 2016-05-04:
Actually, this may have been an issue the the another package, after I did Pkg.update() and recompiled, the julia node began publishing messages on the topic again, using the launch file.
Not sure what exactly caused this behavior, but everything works now

closed 2016-05-04 by jgonzal13