jdlangs/RobotOS.jl #15: Resolve callback crashes in Julia 0.5

phobon, 2016-08-04:
It seems rospy's method of calling the provided callback of a subscriber in a separate thread is not valid with the new multithreading capability of 0.5. Probably some significant reworking is going to be needed according to the suggested work-around and this package won't be working on 0.5 until it gets resolved.

See https://github.com/JuliaLang/julia/issues/17746 and all the related discussion through the layers of connected issues and mailing list posts.

tkelman, 2016-08-17:
Is there a way to make ROS operate in a single-threaded fallback mode? Or does building Julia with `JULIA_THREADS=0` work for now?

jdlangs, 2016-08-17:
I would be very surprised if the ROS devs had allowed for that sort of control. And their level of documentation is such that it's not even clear what threading model is used so I have no idea.

Your other suggestion is setting `JULIA_THREADS` in `Make.inc`, right? It still crashes when built with it set to 0.

tkelman, 2016-08-17:
Have you inquired on the ROS side about this? It will be kind of unfortunate for this not to work at all until some indeterminate future time when https://github.com/JuliaLang/julia/issues/17746 eventually gets resolved. Maybe trying to figure out what the `uv_async_send` suggestion actually means in terms of implementing it here would be the best option?

jdlangs, 2016-08-17:
I'll have some time to work on it at the end of next week, including doing some digging through the ROS code to make sure I have somewhat of an understanding of what they do.

I think I have a basic understanding of what's going on with the `uv_async_send` workaround so I'll spend some time on that too. The main thing I have no idea about right now is how the message sent to the callback is supposed to be transferred to the Julia thread.

nstiurca, 2016-08-29:
This is a very unfortunate regression in Julia v0.5. (Actually, it sounds like maybe this kind of threading wasn't really supposed to work in earlier Julias anyway?)

Regarding @tkelman 's suggestion of single-threaded ROS, it seems `roscpp` provides that functionality. http://wiki.ros.org/roscpp/Overview/Callbacks%20and%20Spinning But wrapping `roscpp` is probably significantly harder than wrapping `rospy`, at least until Cxx.jl matures.

nstiurca, 2016-08-29:
Is it possible wrap the Julia callback functions defined by users of RobotOS.jl so that when the equivalent Python callback is called, it actually funnels callback requests from whatever thread(s) are involved to the main Julia thread?

Something like this:

 - `RobotOS.init()` sets up a global `queue.Queue` (Python threadsafe queue) which will contain pending callbacks received by `rospy.`.
 - `@roscb` macro which which transforms the Julia callback into a Python wrapper which acts the callback for ROS. Using the macro registers the callback with `RobotOS.jl`. The corresponding Python callback gets passed to `Subscriber()`, and when invoked it puts the ROS message and some reference to the Julia callback to the `Queue`.
 - `RobotOS.spinonce()`, called by the user from main Julia thread, gets and processes all items in the Python `Queue` by unwrapping the Julia callback and calling it with the ROS message.
 - `RobotOS.spin()` can wrap `RobotOS.spinonce()` in a trivial `while rospy.ok()` loop or such.

I think this would work because `rospy` only ends up invoking Python code in its callbacks. Assuming the lifetime of the ROS messages is maintained properly across the threads internal to `rospy` and the Julia thread which eventually processes it.

Care has to be taken to deal with getting messages faster than they can be processed. So maybe have separate `Queue`s for each `Subscriber`, and maybe set their maximum sizes to be the same as what the user requests the `queue_size` of `rospy.Subscriber`.

(I hope this makes sense, let me know if it doesn't and I can try to clarify.)

jdlangs, 2016-08-30:
Does this paraphrase accurately capture your suggestion?

When subscribing to a topic, a pure python callback is created and attached to the rospy object which will store the received message in a python queue. Then, a julia function to look through the queue can be directly invoked at regular intervals, at which point the actual julia callbacks will be called, all in the main thread.

Seems like it should work, I just am not sure how that python callback (python wrapper) will be created. Did you have a strategy in mind for that? Also, having to explicitly call a function on the julia side to look through the queue may not be needed if we can set up an asynchronous task and have it start with that call to `uv_async_send` on the python side.

nstiurca, 2016-08-30:
Yes, that sounds about right.

For the Python wrapper, I'm thinking something like

```
def wrapped_callback(jl_handle, q):
    """Wrap a handle to a Julia callback

    jl_handle: some kind of identifier that can be used to
                    call the Julia callback, even if indirectly
    q: a queue.Queue
    """
    def wrapped_callback_impl(msg):
        q.put((jl_handle, msg))
    return wrapped_callback_impl
```

Then in the main thread, have some Julia stuff like so:

```
registered_callbacks = Dict{String, Function}()
q = queue.Queue() # Python Queue

function register_callback(f::Function, args...; kwargs...)
    jl_handle = make_guid() # or something...
    registered_callbacks[jl_handle] = msg -> f(msg, args...; kwargs...)
    jl_handle
end

function spin_once()
    while !q.empty()          # make sure we still eventually break out and return
                                       # even if messages are coming in faster than we
                                       # can process them
        jl_handle, msg = q.get()    # make sure this doesn't block
        registered_callbacks[jl_handle](msg)
    end
end

function roscb(cb::Function, args...; kwargs...)
    jl_handle = register_callback(f, args...; kwargs...)
    wrapped_callback_impl = wrapped_callback(jl_handle, q) # creates the Python closure
                                                                                 # that will act as the rospy callback
    wrapped_callback_impl
end
```

Then you can subscribe with something like `Subscriber{MessageType}("topic", roscb(my_callback, "extra", "args"), queue_size = 10)`. (Assuming of course that you defined something like `function my_callback(msg::MessageType, arg1, arg2)`.) So I guess you don't actually need to have a macro for wrapping the Julia callback with this implementation, just a wrapper function.

nstiurca, 2016-08-30:
And yes, using `uv_async_send` may be better than spinning, but I personally prefer the `spin`/`spin_once` interface available in `roscpp`.

nstiurca, 2016-08-30:
Also, inside my sketch of `spin_once()` above, `msg` is a Python object. It probably needs to be unwrapped into the corresponding Julia object. The unwrapping can maybe happen in the lambda function that's created in `register_callback`, eg parameterize all the functions on `MessageType`, and then creating the lambda ought to be `msg -> f(convort(MessageType, msg), args...; kwargs...)

nstiurca, 2016-08-30:
And both the `register_callback()` and `roscb()` helpers can be moved into the `Subscriber()` constructor to hide everything from the user of `RobotOS.jl`.

jdlangs, 2016-08-30:
@nstiurca that makes sense to use a bit of helper python glue code. Will you have some available time to start putting together the full implementation? I don't think I could have anything ready until late next week.

jdlangs, 2016-08-30:
As a parallel note, it turned out the failure I mentioned with `JULIA_THREADS` as `0` above was just a bug in the makefile. That's now fixed (JuliaLang/julia#18291) and I've confirmed you can get the old working subscriber behavior by disabling threading in the julia build. Apparently it is still unsafe and liable to crash though? At least that workaround is available even if a long term fix is still needed.

nstiurca, 2016-09-20:
I might have some time to look at it this weekend.

Another option may be to completely switch gears and wrap `roscpp` with `Cxx.jl` since `roscpp` gives you control of threading.

nstiurca, 2016-09-21:
Ok, so `Cxx.jl` doesn't support RTTI (run-time type information), and that's used/needed in `libroscpp`. So much for that idea.

PR #19 (2017-01-14): Use async tasks to invoke callbacks in main thread
Adds helper python classes to store received messages/service requests and notify the main julia
thread to invoke the user-specified callback. Fixes #15.
commit 786dd9c5 (2017-01-14): Use async tasks to invoke callbacks in main thread

Adds helper python classes to store received messages/service requests and notify the main julia
thread to invoke the user-specified callback. Fixes #15.
closed 2017-01-14 by jdlangs