jdlangs/RobotOS.jl #16: Move documentation out of the Github README

phobon, 2016-08-04:
into docstrings and an html site using Documenter.jl.

huckl3b3rry87, 2017-07-18:
I started this [here](https://huckl3b3rry87.github.io/RobotOSDocs.jl/latest/). It is basically a copy and paste of your README.md, but I set things up with .travis and Documenter.jl.

jdlangs, 2017-07-18:
Thanks for taking the initiative to get this started! It's been one of those things that never seems important enough to figure out and work on when I have some free time.

I would like to move the docs into this repository as soon as possible so there isn't potential confusion down the road of people using search engines and coming across an old version. Would you be able to open a pull request of just the README translation into Documenter to get started?

huckl3b3rry87, 2017-07-18:
Of course, it took me some time to do this for my package and since I really like your package and you have been helping me I thought I would help out. 

I made the pull request, but I do not have the proper permissions to your repo so, I can't set things up with travis for you. But, now you just need to (deploy your ssh keys)[https://juliadocs.github.io/Documenter.jl/stable/man/hosting/#SSH-Deploy-Keys-1]. I think that should be it, but if you have any trouble please let me know

PR #39 (2017-10-30): Remove documentation from repository README.md
Closes #16.
closed 2017-10-30 by jdlangs