jdlangs/RobotOS.jl #23: startup issues

huckl3b3rry87, 2017-03-19:
I am using Ubuntu 16.04 and Python 2.7.12 and I get this startup issue:
```
               _
   _       _ _(_)_     |  A fresh approach to technical computing
  (_)     | (_) (_)    |  Documentation: http://docs.julialang.org
   _ _   _| |_  __ _   |  Type "?help" for help.
  | | | | | | |/ _` |  |
  | | |_| | | | (_| |  |  Version 0.5.1 (2017-03-05 13:25 UTC)
 _/ |\__'_|_|_|\__'_|  |  Official http://julialang.org/ release
|__/                   |  x86_64-pc-linux-gnu

julia> using RobotOS
ERROR: LoadError: InitError: PyError (:PyImport_ImportModule) <type 'exceptions.ImportError'>
ImportError('No module named rospkg',)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/__init__.py", line 49, in <module>
    from .client import spin, myargv, init_node, \
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/client.py", line 52, in <module>
    import roslib
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslib/__init__.py", line 50, in <module>
    from roslib.launcher import load_manifest
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslib/launcher.py", line 42, in <module>
    import rospkg

 in pyerr_check at /home/febbo/.julia/v0.5/PyCall/src/exception.jl:56 [inlined]
 in pyerr_check at /home/febbo/.julia/v0.5/PyCall/src/exception.jl:61 [inlined]
 in macro expansion at /home/febbo/.julia/v0.5/PyCall/src/exception.jl:81 [inlined]
 in pyimport(::String) at /home/febbo/.julia/v0.5/PyCall/src/PyCall.jl:392
 in __init__() at /home/febbo/.julia/v0.5/RobotOS/src/RobotOS.jl:21
 in include_from_node1(::String) at ./loading.jl:488
 in eval(::Module, ::Any) at ./boot.jl:234
 in require(::Symbol) at ./loading.jl:415
during initialization of module RobotOS
while loading /home/febbo/.julia/v0.5/RobotOS/src/RobotOS.jl, in expression starting on line 38
```
I think that it probably has to do with the version of python that I am using because it was working before I updated some python packages. 

also, I tried:
```
  Pkg.build("RobotOS")
  Pkg.build("PyCall")
```
with no avail

huckl3b3rry87, 2017-03-19:
also, I have done:
```sudo apt-get install python-rospkg```

I am not sure if it moved some ``.py`` files out of my ``./rospy`` directory. Because it worked before the updates

huckl3b3rry87, 2017-03-19:
It turned out to be an issue with the paths. To fix:

**1. in the terminal type:**
``gksu gedit /etc/bash.bashrc``

**2. Then add this to the bottom of the bashrc file:**
``export PYTHONPATH=$PYTHONPATH:/usr/lib/python2.7/dist-packages``

closed 2017-03-19 by huckl3b3rry87

heyucongtom, 2018-03-10:
Thanks @huckl3b3rry87 . That's a really helpful suggestion that saved many hours.