jdlangs/RobotOS.jl #24: question about roscore

huckl3b3rry87, 2017-03-21:
**disclaimer:** I am a ROS nubbie

I am making a [package](https://github.com/huckl3b3rry87/LiDAR.jl) that can run a Gazebo LiDAR model through ``julia`` and I am using ``RobotOS.jl`` to accomplish this task (thank you by the way:) ).

I noticed in the ``.travis.yml`` file there is a ``roscore`` that is started before the script and then killed afterwards. So, in my module if I do a ``using RobotOS``, [as shown here] (https://github.com/huckl3b3rry87/LiDAR.jl/blob/master/src/LiDAR.jl) what does that do in terms of [roscore](http://wiki.ros.org/roscore)? 

If I run ``roscore`` in the terminal and then then I start using my package (which includes a ``using RobotOS``) then I can see both ``rosmaster`` and ``roscore`` tasks (if I use the ``top`` command in Ubuntu).

But if I do not run ``roscore`` beforehand then I do not see these tasks running. 

Basically, I would like to ensure that I set everything up correctly and make sure that I am not starting ``roscore`` if ``RobotOS`` already started it and that I manage communicate through my nodes as I should be.

I apologize if this is confusing, maybe some basic instructions about using this package in conjunction with ``roscore`` would help address my questions?

Thanks!

jdlangs, 2017-03-22:
Hi @huckl3b3rry87, to directly answer your question, `RobotOS` doesn't do anything on its own with `roscore`. The travis file spawns a `roscore` instance because the testing isn't possible without a ROS master running.

Looking at your example, you seem to trying to start up all the other ROS functionality from within the Julia code, which is odd. The normal thing would be to write a [launch file](http://wiki.ros.org/roslaunch) to start up the system, and your Julia package would be treated just like a normal python node. Launch files are also nice because they auto-start a ROS master if one isn't already running.

huckl3b3rry87, 2017-03-22:
@jdlangs thank you for your response, it is very helpful.

The reason that I started everything in julia is because I want julia to drive the simulation. I have several other things that will be simulated, including a vehicle model in MATLAB and a large optimization written in julia. So, I was hoping to develop API for the gazebo LiDAR model so I can call it in julia. Since the LiDAR is a simulation, I realize that I would have to start it up and then update the parameters (basically giving it a trajectory to follow for a particular time interval) and then the output of the model would be a sequence of point cloud data during that time interval. So, it would always be running in the background, it would just be waiting for julia for new information. 

Basically, I have a big simulation that is already managed by julia code and I was hoping to keep it that way.

With this said, I want to reiterate that I am a ROS newbie and I am also a python newbie, my background is in MATLAB and now julia. So, perhaps I should be thinking about this differently?

Thinking about this more and reading more about RobotOS and ROS. It seems that I sort of missed the boat in terms of understanding the purpose of RobotOS. Correct me if I am wrong, but it seems that RobotOS was never designed to allow julia to be in command of the simulation, but to allow for julia code to be one a ROS node that is managed by a ROS master.

closed 2017-03-22 by huckl3b3rry87

jdlangs, 2017-03-23:
Hope you're able to come up with something cool! Just think about `RobotOS` as a direct alternative to `roscpp` and `rospy` and everything else is business as usual with ROS.