jdlangs/RobotOS.jl #30: issue publishing to a Gazebo node

huckl3b3rry87, 2017-07-18:
I made a rough example [here](https://huckl3b3rry87.github.io/RobotOSDocs.jl/latest/tutorials/Gazebo/main.html) of my issue. At the bottom it of this page the error that I am getting is shown.

jdlangs, 2017-07-18:
These are very basic errors you are running into and I would encourage you to spend more time trying to understand what's happening on your own before you come to the internet for help.

In this case, when Julia gives you a `MethodError`, it means you tried to call a function with incorrect argument types. If you review the documentation on services (both here and in `rospy`), you should be able to easily find the correct way to invoke a `ServiceProxy{gazebo_msgs.srv.SetModelState}` object.

closed 2017-07-18 by jdlangs

huckl3b3rry87, 2017-07-18:
I did spend a bunch of time trying to figure this out on my own, eventually I found this [example](https://github.com/ipa320/srs_public/blob/master/srs_user_tests/ros/scripts/move_robot_sim.py#L44) where they define:
```
 g_set_state = rospy.ServiceProxy("/gazebo/set_model_state",SetModelState)
```

Then later they invoke the object (with something like):
```
 state = ModelState()
 ret = g_set_state(state)
```
I have also done this in python with no problem. 

I understand that they are different types ``SetModelState`` and ``ModelState``, but from what I can tell you invoke a service call to the ``/gazebo/set_model_state`` service using a ``ModelState`` type not a ``SetModelState`` type.

jdlangs, 2017-07-19:
This actually is a bit subtle, but I think you would have found the explanation had you first focused on how to successfully call the service in Julia instead of assuming ROS python code would work equally well in Julia.

Hopefully, you've seen explained in the docs that when you import the `SetModelState` service, it autogenerates `SetModelStateRequest` and `SetModelStateResponse` types for you. You need to send a `SetModelStateRequest` object when calling the `ServiceProxy`. In rospy, a [shortcut is provided](http://wiki.ros.org/rospy/Overview/Services#Calling_services) so that instead of creating the request object, you can just directly send the request message contents. In the case of `SetModelStateRequest`, that's just a `ModelState` object. Maybe it's a good idea to add a note in the docs that this python shortcut is not available (currently) through RobotOS.

huckl3b3rry87, 2017-07-19:
@jdlangs thank you very much for your help with this. I don't mean to ask too many questions. I know that you covered that in the docs, it was something that I read a bunch of times but for some reason it did not sync in that that is what those types are for and how to use them, it it just very difficult for me to learn without examples.