jdlangs/RobotOS.jl #33: support for tf

huckl3b3rry87, 2017-07-27:
I was having some trouble importing some for the functionality for [tf](https://w3.cs.jmu.edu/spragunr/CS354_S14/labs/tf_lab/html/tf-module.html) for instance:
```
julia> @rosimport tf.transformations: LookupTransform
ERROR: AssertionError: Improper @rosimport input
```
When I [noticed here](http://wiki.ros.org/rospy/Overview) that ``tf``` not in in ``rospy`` itself. Is this expected? Or am I making another silly error?

Thank you

jdlangs, 2017-07-28:
Yeah, unfortunately `tf` is a completely separate python library and I don't do anything to convert it to Julia. What you can do is think of it as a normal python package that you use `PyCall.jl` to interact with (i.e., using `@pyimport`). The only snag will be the `transformPointCloud`, `transformPose`, etc... functions which ask for ROS message types as arguments. To make it work, you'll have to call `convert(PyCall.PyObject, my_msg)` to send it as a python object. This happens when publishing and subscribing to Julia messages through rospy; it just is hidden from the user.

jdlangs, 2017-07-28:
I'll close this issue since it wouldn't be appropriate to put wrappers around specific ROS packages in a client library. Possibly you may be interested in writing your own package that specifically wraps `tf` for use in Julia and hides all the python interface from the user.

closed 2017-07-28 by jdlangs