jdlangs/RobotOS.jl #34: interfacing with custom message

huckl3b3rry87, 2017-08-01:
Similar to [this tutorial](http://wiki.ros.org/ROS/Tutorials/CustomMessagePublisherSubscriber%28python%29) I would like to interface (publish and subscribe) with some custom message types using RobotOS.jl. Is this possible?

Thanks

jdlangs, 2017-08-01:
Yes, definitely. Is there something you've tried along those lines that doesn't work?

huckl3b3rry87, 2017-08-01:
Since the package was not an python package, I assumed that the ``@rosimport`` command would not apply, but it does. Here is an example:
```
febbo@febbo-HP-ZBook-17-G2:~$ julia
               _
   _       _ _(_)_     |  A fresh approach to technical computing
  (_)     | (_) (_)    |  Documentation: https://docs.julialang.org
   _ _   _| |_  __ _   |  Type "?help" for help.
  | | | | | | |/ _` |  |
  | | |_| | | | (_| |  |  Version 0.6.0 (2017-06-19 13:05 UTC)
 _/ |\__'_|_|_|\__'_|  |  Official http://julialang.org/ release
|__/                   |  x86_64-pc-linux-gnu

julia> using RobotOS
julia> @rosimport obstacle_detector.msg: Obstacles
julia> rostypegen()
julia> using obstacle_detector.msg
julia> Obstacles()
obstacle_detector.msg.Obstacles(std_msgs.msg.Header(0x00000000, RobotOS.Time(0, 0), ""), obstacle_detector.msg.SegmentObstacle[], obstacle_detector.msg.CircleObstacle[])
```

closed 2017-08-01 by huckl3b3rry87

jdlangs, 2017-08-03:
I'm not sure what you mean by "make your package into a python package". Any package that defines it's own message types will generate both C++ and python files for the message classes automatically when built. As long as that package is in your currently active workspace when you run julia, the python path will include that package and `@rosimport` will find those generated classes.

jdlangs, 2017-08-03:
If you weren't aware of it already, use `catkin_create_pkg` to generate your own ROS packages. The generated `CMakeLists.txt` file has very clear directions on what to change to make use of various features.