jdlangs/RobotOS.jl #45: precompile issue

huckl3b3rry87, 2018-06-13:
@jdlangs I have been trying to figure out an issue (that I created with https://github.com/jdlangs/RobotOS.jl/commit/4c65224c9774db872c6b247dba09536deda8d8fd) for some time now with no luck, it is an issue in my package
https://github.com/JuliaMPC/MAVs/issues/28

Basically, from a 
```
using RobotOS
```
I get
```
ERROR: LoadError: Module RobotOS declares __precompile__(true) but require failed to create a usable precompiled cache file.
Stacktrace:Waiting for '/gazebo/unpause_physics' service...
 [1] 
_requirelidar simulation in Gazebo has been initialized
(::Symbol) at ./loading.jl:502
 [2] require(::Symbol) at ./loading.jl:405
 [3] include_from_node1(::String) at ./loading.jl:576
 [4] include(::String) at ./sysimg.jl:14
 [5] process_options(::Base.JLOptions) at ./client.jl:305
 [6] _start() at ./client.jl:371
while loading /home/mavs/MAVs/ros/src/models/gazebo/vehicle_description/link_positions.jl, in expression starting on line 2
```
https://github.com/JuliaMPC/MAVs/blob/master/ros/src/models/gazebo/vehicle_description/link_positions.jl#L2

To fix it, I have been just deleting the ``.julia\.cache`` folder, but now I am creating a Docker image for my software and I do not want users to have to deal with this. Do you have any idea what may be causing this?

Thanks!

jdlangs, 2018-06-14:
Not sure what might be happening there. Can you run `Base.compilecache("RobotOS")` after deleting the cache file to see if a more informative error is thrown? And I assume you're running julia 0.6?

huckl3b3rry87, 2018-06-19:
@jdlangs sorry for the delay.

When I just run it to try and reproduce the error:
```
mavs@febbo-HP-ZBook-17-G2:~$ julia
               _
   _       _ _(_)_     |  A fresh approach to technical computing
  (_)     | (_) (_)    |  Documentation: https://docs.julialang.org
   _ _   _| |_  __ _   |  Type "?help" for help.
  | | | | | | |/ _` |  |
  | | |_| | | | (_| |  |  Version 0.6.2 (2017-12-13 18:08 UTC)
 _/ |\__'_|_|_|\__'_|  |  Official http://julialang.org/ release
|__/                   |  x86_64-pc-linux-gnu

julia> using RobotOS

julia> Base.compilecache("RobotOS")
INFO: Recompiling stale cache file /home/mavs/.julia/lib/v0.6/RobotOS.ji for module RobotOS.
WARNING: Module PyCall with uuid 17207982653232 is missing from the cache.
This may mean module PyCall does not support precompilation but is imported by a module that does.
ERROR: LoadError: Declaring __precompile__(false) is not allowed in files that are being precompiled.
Stacktrace:
 [1] _require(::Symbol) at ./loading.jl:455
 [2] require(::Symbol) at ./loading.jl:405
 [3] include_from_node1(::String) at ./loading.jl:576
 [4] include(::String) at ./sysimg.jl:14
 [5] anonymous at ./<missing>:2
while loading /home/mavs/.julia/v0.6/RobotOS/src/RobotOS.jl, in expression starting on line 5
ERROR: Failed to precompile RobotOS to /home/mavs/.julia/lib/v0.6/RobotOS.ji.
Stacktrace:
 [1] compilecache(::String) at ./loading.jl:710

```

If I first remove the cache file (as you asked)  I do not get the error:
```
mavs@febbo-HP-ZBook-17-G2:~$ rm -rf /home/mavs/.julia/.cache
mavs@febbo-HP-ZBook-17-G2:~$ julia
               _
   _       _ _(_)_     |  A fresh approach to technical computing
  (_)     | (_) (_)    |  Documentation: https://docs.julialang.org
   _ _   _| |_  __ _   |  Type "?help" for help.
  | | | | | | |/ _` |  |
  | | |_| | | | (_| |  |  Version 0.6.2 (2017-12-13 18:08 UTC)
 _/ |\__'_|_|_|\__'_|  |  Official http://julialang.org/ release
|__/                   |  x86_64-pc-linux-gnu

julia> Base.compilecache("RobotOS")
INFO: Precompiling module RobotOS.
"/home/mavs/.julia/lib/v0.6/RobotOS.ji"
```
Seems to be an issue with PyCall.jl?

jdlangs, 2018-06-24:
It does look like it's probably an upstream issue. Feel free to reopen this if you discover otherwise.

closed 2018-06-24 by jdlangs

huckl3b3rry87, 2018-09-15:
created an issue in PyCall
https://github.com/JuliaPy/PyCall.jl/issues/568