jdlangs/RobotOS.jl #49: Precompilation Examples with RobotOS

govvijaycal, 2018-07-15:
Hi,

I was wondering if it is possible to precompile Julia code with ROS publishers/subscribers.  I would like to simply import a module and have the __init__ function call a main function to start a publisher loop and subscriber callbacks.

The issue I'm running into is that the message types are not defined until rostypegen() is called.  So I'm not sure if you can set up the publisher/subscriber objects or calls to the RobotOS functions ahead of runtime.

Are there any examples of doing this type of precompilation?  This would help me to reduce start up time in running my ROS julia node.

Thanks for the help!

schmrlng, 2018-07-17:
With PR #47, something like this might work for your use case:

```julia
__precompile__()

module ROSDemo

using RobotOS

@rosimport std_msgs.msg: String
rostypegen(current_module())
import .std_msgs.msg: StringMsg

function reverse_me_callback(msg::StringMsg, doubled_pub)
    publish(doubled_pub, StringMsg(reverse(msg.data)))
end

function __init__()
    @rosimport std_msgs.msg: String
    init_node("reverse_node")
    doubled_pub = Publisher{StringMsg}("/reversed", queue_size=10)
    Subscriber{StringMsg}("/reverse_me", reverse_me_callback, (doubled_pub,), queue_size=10)
    @spawn spin()
end

end # module
```

For a demo of the above, run these commands in 3 separate terminals:
`julia -e "using ROSDemo"`
`rostopic echo /reversed`
`rostopic pub -1 /reverse_me std_msgs/String "data: hello"`

schmrlng, 2018-07-17:
That's probably worth a shot - I agree that only the changes from #47 in `src/gentypes.jl` should be necessary, but it's possible that julia AST/metaprogramming syntax has changed enough between 0.4 and 0.6 that some additional fiddling around might be necessary. At face value, however, I can't think of why the content of this change wouldn't be backward compatible, if not the actual code diff itself.

govvijaycal, 2018-08-04:
I finally got around to testing this on Julia 0.6.4 and it seemed to load pretty quickly and work without issues.  Thanks for the help!

closed 2018-08-04 by govvijaycal