jdlangs/RobotOS.jl #52: using a module that uses RobotOS.jl

gstavrinos, 2018-09-09:
Hey,

I am developing a julia module that uses RobotOS.jl internally. When I am running my module as a standalone script everything is working fine, but when I include my module via the `using` keyword, the messages (geometry_msgs) fail to compile. This problem is not present if I use the `include` keyword. Any thoughts?

Thanks,

George

tradeA, 2018-09-10:
by "standalone script" what so you mean exactly? As far as I can tell, you have to be in the scripts directory of a ros package. You also have to make sure that you pythonpath is set to same python instance as the ros system

gstavrinos, 2018-09-11:
Hello tradeA,

My ROS package is this https://github.com/gstavrinos/ez_pick_and_place.

In the src directory is the to-be module, and in the example directory is the .jl file that should use it.

Currently I am using `include` in order to override this problem, but I would prefer to be able to use `using`. 

(Also, keep in my mind that I am using the version that is currently in #48 in order to support julia-1.0.)

Thanks for the interest,

George

jdlangs, 2018-09-21:
It sounds like the module precompilation issue that was addressed by #47. Have you tried generating the message modules inside your package module instead of `Main`?

gstavrinos, 2018-09-21:
I tried it, but couldn't work it out. The way I bypassed it was by using `include`.