jdlangs/RobotOS.jl #53: Cannot publish any kind of message

gstavrinos, 2018-09-14:
While working on a pick and place wrapper for MoveIt, I noticed that my messages were not published. At first I thought that there was a problem with my code or MoveIt, but I created a simple node and still cannot publish a message. My (simple) node is the following:

`
#!/usr/bin/env julia
using RobotOS
@rosimport geometry_msgs.msg: PoseStamped
rostypegen()
using geometry_msgs.msg
init_node("julia_test")
t = Publisher("/test_topic", PoseStamped, queue_size = 1)
sleep(5)
ps = PoseStamped()
ps.header.stamp = RobotOS.now()

i = 1

while i < 10
    publish(t, ps)
    println("Published"*string(i))
    sleep(1)
    i += 1
end
`

Any thoughts?

jdlangs, 2018-09-21:
I think your problem is you are calling Julia's `sleep` which is not allowing rospy to do the message publishing. Try using `rossleep` instead or calling `sleep` with a `Duration` instance.

gstavrinos, 2018-09-21:
Just tried it, but the problem persists. I will wait for the julia-1.0 compatibility update, and update the issue here.

gstavrinos, 2018-10-12:
Latest version of RobotOS  with Julia-1.0.1 seems to be working. I can publish messages normally. The issue #57 regarding services, persists.

Thanks for the update, Josh! I am closing this.

closed 2018-10-12 by gstavrinos