jdlangs/RobotOS.jl #54: python path issue

huckl3b3rry87, 2018-09-15:
I am running into this issue
```
   _       _ _(_)_     |  A fresh approach to technical computing
  (_)     | (_) (_)    |  Documentation: https://docs.julialang.org
   _ _   _| |_  __ _   |  Type "?help" for help.
  | | | | | | |/ _` |  |
  | | |_| | | | (_| |  |  Version 0.6.2 (2017-12-13 18:08 UTC)
 _/ |\__'_|_|_|\__'_|  |  Official http://julialang.org/ release
|__/                   |  x86_64-pc-linux-gnu

julia> using RobotOS
ERROR: InitError: PyError (PyImport_ImportModule

The Python package rospy could not be found by pyimport. Usually this means
that you did not install rospy in the Python version being used by PyCall.

PyCall is currently configured to use the Python version at:

/usr/bin/python3

and you should use whatever mechanism you usually use (apt-get, pip, conda,
etcetera) to install the Python package containing the rospy module.

One alternative is to re-configure PyCall to use a different Python
version on your system: set ENV["PYTHON"] to the path/name of the python
executable you want to use, run Pkg.build("PyCall"), and re-launch Julia.

Another alternative is to configure PyCall to use a Julia-specific Python
distribution via the Conda.jl package (which installs a private Anaconda
Python distribution), which has the advantage that packages can be installed
and kept up-to-date via Julia.  As explained in the PyCall documentation,
set ENV["PYTHON"]="", run Pkg.build("PyCall"), and re-launch Julia. Then,
To install the rospy module, you can use `pyimport_conda("rospy", PKG)`,
where PKG is the Anaconda package the contains the module rospy,
or alternatively you can use the Conda package directly (via
`using Conda` followed by `Conda.add` etcetera).

) <class 'ImportError'>
ImportError("No module named 'yaml'",)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/__init__.py", line 47, in <module>
    from std_msgs.msg import Header
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/std_msgs/msg/__init__.py", line 1, in <module>
    from ._Bool import *
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/std_msgs/msg/_Bool.py", line 5, in <module>
    import genpy
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/genpy/__init__.py", line 34, in <module>
    from . message import Message, SerializationError, DeserializationError, MessageException, struct_I
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/genpy/message.py", line 44, in <module>
    import yaml

Stacktrace:
 [1] pyimport(::String) at /home/mavs/.julia/v0.6/PyCall/src/PyCall.jl:474
 [2] __init__() at /home/mavs/.julia/v0.6/RobotOS/src/RobotOS.jl:24
 [3] _include_from_serialized(::String) at ./loading.jl:157
 [4] _require_from_serialized(::Int64, ::Symbol, ::String, ::Bool) at ./loading.jl:200
 [5] _require_search_from_serialized(::Int64, ::Symbol, ::String, ::Bool) at ./loading.jl:236
 [6] _require(::Symbol) at ./loading.jl:441
 [7] require(::Symbol) at ./loading.jl:405
during initialization of module RobotOS
```

huckl3b3rry87, 2018-09-15:
I had to do 
```
sudo apt-get install python-yaml
```
then in julia
```
ENV["PYTHON"]="/usr/bin/python2.7"
Pkg.build("PyCall")
```

closed 2018-09-15 by huckl3b3rry87

mahtaao, 2020-08-24:
How does `pyyaml `help exactly? You installed it on `python2.7` and then?

mahtaao, 2020-08-24:
> 
> 
> I had to do
> 
> ```
> sudo apt-get install python-yaml
> ```
> 
> then in julia
> 
> ```
> ENV["PYTHON"]="/usr/bin/python2.7"
> Pkg.build("PyCall")
> ```

How does `pyyaml` help exactly? You installed it on `python2.7` and then?

PavanproJack, 2020-10-07:
> I had to do
> 
> ```
> sudo apt-get install python-yaml
> ```
> 
> then in julia
> 
> ```
> ENV["PYTHON"]="/usr/bin/python2.7"
> Pkg.build("PyCall")
> ```

This works. Thank you