jdlangs/RobotOS.jl #60: julia-1.0 and julia-0.7 can not find ros msg?

Aidous, 2018-11-26:
ERROR: LoadError: ArgumentError: Package geometry_msgs not found in current path:
- Run `Pkg.add("geometry_msgs")` to install the geometry_msgs package.

Stacktrace:
 [1] require(::Module, ::Symbol) at ./loading.jl:817
 [2] include at ./boot.jl:317 [inlined]
 [3] include_relative(::Module, ::String) at ./loading.jl:1038
 [4] include(::Module, ::String) at ./sysimg.jl:29
 [5] exec_options(::Base.JLOptions) at ./client.jl:239
 [6] _start() at ./client.jl:432

gstavrinos, 2018-11-27:
The Julia wrapper uses python internally, so you can't add a new ros package like this. Just use your package manager normally, and then just import them in Julia like specified in the documentation.

gstavrinos, 2018-11-28:
Oh, ok. The try this:

Instead of `using geometry_msgs.msg` try `using .geometry_msgs.msg`. Good luck!

Aidous, 2018-11-28:
> Oh, ok. Then maybe try this:
> 
> Instead of `using geometry_msgs.msg` try `using .geometry_msgs.msg`. Good luck!

Yes, it works! Thanks very much!

closed 2018-11-28 by Aidous