jdlangs/RobotOS.jl #72: Quiet error when generating custom messages

AndrewJSchoen, 2019-08-13:
I have some custom messages, and was getting a quiet error when formulating them in a certain way. In `custom.msg`, formulating with `std_msgs/Float32[]` seems to generate, except that subscriber callbacks which accept it silently fail. 

```
# custom.msg
std_msgs/Header header
std_msgs/Float32[] dc_values
```

```
# Subscriber
function custom_cb(data::custom)
    loginfo(data)
end

init_node("test_node")
Subscriber{custom}("/test", custom_cb)
loop_rate = Rate(1)
while !is_shutdown()
    rossleep(loop_rate)
end
```

Nothing logs or errors on the subscriber.

Took me a bit to find the issue, but when I switch the message generation to the following:
```
# custom.msg
std_msgs/Header header
float32[] dc_values
```
Everything works as expected.

jdlangs, 2019-08-19:
It's likely an error around having to rename the `Float32` message type to `Float32Msg`. Is an error thrown when you attempt to instantiate a `custom` message after it gets generated?

AndrewJSchoen, 2019-08-19:
There are no errors thrown.

jdlangs, 2019-10-14:
I took a look at this and was unable to reproduce any problem based on the code sample provided. If this is still a problem please provide a full standalone package/workspace that causes it. Otherwise, I'll plan to close this issue in some time.

closed 2019-11-18 by jdlangs