jdlangs/RobotOS.jl #73: Issue with using PyCall.jl along with RobotOS.jl

haruki-msl, 2019-10-11:
I have recently encountered an issue where PyCall returns IndexError after importing RobotOS.

```
   _       _ _(_)_     |  Documentation: https://docs.julialang.org
  (_)     | (_) (_)    |
   _ _   _| |_  __ _   |  Type "?" for help, "]?" for Pkg help.
  | | | | | | |/ _` |  |
  | | |_| | | | (_| |  |  Version 1.2.0 (2019-08-20)
 _/ |\__'_|_|_|\__'_|  |  Official https://julialang.org/ release
|__/                   |

julia> using PyCall

julia> py"""
       import argparse
       parser = argparse.ArgumentParser()
       """

julia> import RobotOS

julia> py"""
       import argparse
       parser = argparse.ArgumentParser()
       """
ERROR: PyError ($(Expr(:escape, :(ccall(#= /home/haruki/.julia/packages/PyCall/ttONZ/src/pyeval.jl:39 =# @pysym(:PyEval_EvalCode), PyPtr, (PyPtr, PyPtr, PyPtr), o, globals, locals))))) <type 'exceptions.IndexError'>
IndexError('list index out of range',)
  File "/home/haruki/.julia/packages/PyCall/ttONZ/src/pyeval.jl", line 2, in <module>
    const Py_file_input = 257
  File "/usr/lib/python2.7/argparse.py", line 1586, in __init__
    prog = _os.path.basename(_sys.argv[0])
```

This does not occur with other packages that depend on PyCall, for example, PyPlot.
```
julia> using PyCall

julia> py"""
       import argparse
       parser = argparse.ArgumentParser()
       """

julia> import PyPlot

julia> py"""
       import argparse
       parser = argparse.ArgumentParser()
       """

julia> 
```

I wonder if this is a PyCall issue, a RobotOS issue, or a problem specific to my environment.

Julia Version = v1.2.0
PyCall.pyversion = v2.7.12

It also happened with PyCall.pyversion = v3.6.9.

haruki-msl, 2019-10-11:
Actually I've just found that sys.argv is replaced by Julia's ARGS, which changes sys.argv from `PyObject ['']` to `PyObject[]`. 
https://github.com/jdlangs/RobotOS.jl/blob/master/src/RobotOS.jl#L21

This is causing the issue, maybe?

jdlangs, 2019-10-14:
Yes, that was the problem. Thanks for reporting the problem and looking into it!

PR #74 (2019-10-14): Only overwrite `sys.argv` if we have ARGS
Fixes #73
closed 2019-10-14 by jdlangs