jdlangs/RobotOS.jl #80: RSO2 support

AndreasAZiegler, 2020-04-29:
As a robotics engineer, I’m working with ROS2. From an (software) engineering and research perspective, I’m quite interested in Julia however with the lack of a ROS2 client library for Julia, it’s for me currently a bit difficult to start using Julia. Is there the plan to support ROS2?

jdlangs, 2020-05-03:
Hi Andreas, the short answer is yes, I want this package to support ROS2. I also use ROS2 at work so I have some incentive to work on it. I did do some initial prototyping work a few months ago to experiment with using librcl directly and to do Julia type generation. The basics for all that has been worked out but I haven't yet been able to get back to fleshing it out properly.

At this point, with the covid disruption, it's hard to say when I can get back into working on this package. I'm expecting to have a bit less workload come mid-summer so that might be the timeline.

ChetanVardhan, 2020-09-03:
Any chance that ROS2 will be implemented via Cxx instead of PyCall, as was in the case of RobotOS.jl?
Is there any update on "ROS2 on Julia"?

JinraeKim, 2020-10-20:
I'm really looking forward to use ROS2 in Julia as a newbie of both ROS2 and Julia! 😄

clad26, 2020-12-10:
Any update on this?

lassepe, 2021-08-07:
@jdlangs you said you already started playing around with wrapping librcl in Julia. Do have a branch or a repo with your initial sketch? I'd be interested in doing some more poking.

jdlangs, 2021-08-10:
@lassepe thanks for your interest here. I'm happy to make my initial work public and you can now find it on the [ros2 branch](https://github.com/jdlangs/RobotOS.jl/tree/ros2). As usual, I don't expect to have much bandwidth to continue working on this currently but maybe this starting point will inspire some further efforts.