jdlangs/RobotOS.jl #81: Test problem

pauljurczak, 2020-05-31:
I'm following instructions to test `RobotOS` package on Ubuntu 18.04 with Python 2.7 and I'm getting
```
(@v1.4) pkg> test RobotOS
    Testing RobotOS
Status `/tmp/jl_XxrPsx/Manifest.toml`
  [8f4d0f93] Conda v1.4.1
  [682c06a0] JSON v0.21.0
  [1914dd2f] MacroTools v0.5.5
  [69de0a69] Parsers v1.0.4
  [438e738f] PyCall v1.91.4
  [22415677] RobotOS v0.7.1
  [81def892] VersionParsing v1.2.0
  [2a0f44e3] Base64 
  [ade2ca70] Dates 
  [8ba89e20] Distributed 
  [b77e0a4c] InteractiveUtils 
  [8f399da3] Libdl 
  [37e2e46d] LinearAlgebra 
  [56ddb016] Logging 
  [d6f4376e] Markdown 
  [a63ad114] Mmap 
  [de0858da] Printf 
  [9a3f8284] Random 
  [9e88b42a] Serialization 
  [6462fe0b] Sockets 
  [8dfed614] Test 
  [4ec0a83e] Unicode 
[INFO] [1590897901.297812]: testing: info
[WARN] [1590897901.302845]: testing: warn
[ERROR] [1590897901.308276]: testing: err
[FATAL] [1590897901.313352]: testing: fatal
Importing new package: geometry_msgs.msg
Message type import: geometry_msgs.msg.PoseStamped
Importing python package: geometry_msgs.msg
Dependency: std_msgs/Header
    Creating new package: std_msgs
    Message type import: std_msgs.msg.Header
    Importing python package: std_msgs.msg
Dependency: geometry_msgs/Pose
    Message type import: geometry_msgs.msg.Pose
    Dependency: geometry_msgs/Point
        Message type import: geometry_msgs.msg.Point
    Dependency: geometry_msgs/Quaternion
        Message type import: geometry_msgs.msg.Quaternion
Message type import: geometry_msgs.msg.Vector3
Importing new package: visualization_msgs.msg
Message type import: visualization_msgs.msg.Marker
Importing python package: visualization_msgs.msg
Dependency: std_msgs/Header
Dependency: geometry_msgs/Pose
Dependency: geometry_msgs/Vector3
Dependency: std_msgs/ColorRGBA
    Message type import: std_msgs.msg.ColorRGBA
Dependency: geometry_msgs/Point[]
Dependency: std_msgs/ColorRGBA[]
Importing new package: std_srvs.srv
Service type import: std_srvs.srv.Empty
Importing python package: std_srvs.srv
Service type import: std_srvs.srv.SetBool
Importing new package: nav_msgs.srv
Service type import: nav_msgs.srv.GetPlan
Importing python package: nav_msgs.srv
Dependency: geometry_msgs/PoseStamped
Dependency: geometry_msgs/PoseStamped
Dependency: nav_msgs/Path
    Message type import: nav_msgs.msg.Path
    Importing python package: nav_msgs.msg
    Dependency: std_msgs/Header
    Dependency: geometry_msgs/PoseStamped[]
Message type import: std_msgs.msg.Empty
Message type import: std_msgs.msg.Float64
┌ Warning: Message type 'Float64' conflicts with Julia builtin, will be imported as 'Float64Msg'
└ @ RobotOS ~/.julia/packages/RobotOS/OnyF4/src/gentypes.jl:181
Message type import: std_msgs.msg.String
┌ Warning: Message type 'String' conflicts with Julia builtin, will be imported as 'StringMsg'
└ @ RobotOS ~/.julia/packages/RobotOS/OnyF4/src/gentypes.jl:181
Importing new package: fake_msgs.msg
Message type import: fake_msgs.msg.FakeMsg
Importing python package: fake_msgs.msg
PyError (PyImport_ImportModule

The Python package fake_msgs.msg could not be found by pyimport. Usually this means
that you did not install fake_msgs.msg in the Python version being used by PyCall.
```

I don't know how to `install the Python package containing the fake_msgs.msg module`. What is the name of this package?

pauljurczak, 2020-05-31:
I also noticed that one of the processes required for the test, fails:
```
paul@paul-desktop:~/.julia/packages/RobotOS/OnyF4/test$ ./echonode.py 
import-im6.q16: not authorized `rospy' @ error/constitute.c/WriteImage/1037.
from: can't read /var/mail/geometry_msgs.msg
from: can't read /var/mail/std_srvs.srv
from: can't read /var/mail/nav_msgs.srv
./echonode.py: line 10: syntax error near unexpected token `('
./echonode.py: line 10: `class Echo(object):'
```

jdlangs, 2020-06-05:
Hi @pauljurczak,

In the first comment, the package is actually testing that the system errors when you ask to create a message type that doesn't exist (`fake_msgs.msg.FakeMsg`), so that is working properly.

For the second comment, I'm not sure what that error is, but since it happens when running a normal python ROS node it is probably an issue with your system in particular and not something to do with Julia or this package.

Thanks!

closed 2020-06-05 by jdlangs