jdlangs/RobotOS.jl #82: Julia 1.5.0 support

huckl3b3rry87, 2020-08-18:
Hi, 

I am hitting this error 
```
Error: LoadError: type LineNumberNode has no field args
```
line 520 in gentypes

It looks like it was a common error when people moved to Julia 0.7, but I am not sure the fix.

Also, side note, should RobotOS.jl work on melodic?

huckl3b3rry87, 2020-08-18:
https://github.com/jdlangs/RobotOS.jl/pull/83/commits

huckl3b3rry87, 2020-08-18:
apparently travis did not get triggered with my build, so I just set it up and tested on my local machine
```
febbo@febbo:~/.julia/dev/RobotOS/test$ julia runtests.jl 
[INFO] [1597769199.185428]: testing: info
[WARN] [1597769199.191713]: testing: warn
[ERROR] [1597769199.198581]: testing: err
[FATAL] [1597769199.205693]: testing: fatal
Importing new package: geometry_msgs.msg
Message type import: geometry_msgs.msg.PoseStamped
Importing python package: geometry_msgs.msg
Dependency: std_msgs/Header
	Creating new package: std_msgs
	Message type import: std_msgs.msg.Header
	Importing python package: std_msgs.msg
Dependency: geometry_msgs/Pose
	Message type import: geometry_msgs.msg.Pose
	Dependency: geometry_msgs/Point
		Message type import: geometry_msgs.msg.Point
	Dependency: geometry_msgs/Quaternion
		Message type import: geometry_msgs.msg.Quaternion
Message type import: geometry_msgs.msg.Vector3
Importing new package: visualization_msgs.msg
Message type import: visualization_msgs.msg.Marker
Importing python package: visualization_msgs.msg
Dependency: std_msgs/Header
Dependency: geometry_msgs/Pose
Dependency: geometry_msgs/Vector3
Dependency: std_msgs/ColorRGBA
	Message type import: std_msgs.msg.ColorRGBA
Dependency: geometry_msgs/Point[]
Dependency: std_msgs/ColorRGBA[]
Importing new package: std_srvs.srv
Service type import: std_srvs.srv.Empty
Importing python package: std_srvs.srv
Service type import: std_srvs.srv.SetBool
Importing new package: nav_msgs.srv
Service type import: nav_msgs.srv.GetPlan
Importing python package: nav_msgs.srv
Dependency: geometry_msgs/PoseStamped
Dependency: geometry_msgs/PoseStamped
Dependency: nav_msgs/Path
	Message type import: nav_msgs.msg.Path
	Importing python package: nav_msgs.msg
	Dependency: std_msgs/Header
	Dependency: geometry_msgs/PoseStamped[]
Message type import: std_msgs.msg.Empty
Message type import: std_msgs.msg.Float64
┌ Warning: Message type 'Float64' conflicts with Julia builtin, will be imported as 'Float64Msg'
└ @ RobotOS ~/.julia/dev/RobotOS/src/gentypes.jl:181
Message type import: std_msgs.msg.String
┌ Warning: Message type 'String' conflicts with Julia builtin, will be imported as 'StringMsg'
└ @ RobotOS ~/.julia/dev/RobotOS/src/gentypes.jl:181
Importing new package: fake_msgs.msg
Message type import: fake_msgs.msg.FakeMsg
Importing python package: fake_msgs.msg
PyError (PyImport_ImportModule

The Python package fake_msgs.msg could not be found by pyimport. Usually this means
that you did not install fake_msgs.msg in the Python version being used by PyCall.

PyCall is currently configured to use the Python version at:

/usr/bin/python2.7

and you should use whatever mechanism you usually use (apt-get, pip, conda,
etcetera) to install the Python package containing the fake_msgs.msg module.

One alternative is to re-configure PyCall to use a different Python
version on your system: set ENV["PYTHON"] to the path/name of the python
executable you want to use, run Pkg.build("PyCall"), and re-launch Julia.

Another alternative is to configure PyCall to use a Julia-specific Python
distribution via the Conda.jl package (which installs a private Anaconda
Python distribution), which has the advantage that packages can be installed
and kept up-to-date via Julia.  As explained in the PyCall documentation,
set ENV["PYTHON"]="", run Pkg.build("PyCall"), and re-launch Julia. Then,
To install the fake_msgs.msg module, you can use `pyimport_conda("fake_msgs.msg", PKG)`,
where PKG is the Anaconda package the contains the module fake_msgs.msg,
or alternatively you can use the Conda package directly (via
`using Conda` followed by `Conda.add` etcetera).

) <type 'exceptions.ImportError'>
ImportError('No module named fake_msgs.msg',)
Message type import: std_msgs.msg.FakeMsg
Service type import: nav_msgs.srv.GetPlanRequest
Building package: std_srvs
	submodule: std_srvs.srv
		Type: EmptyRequest
		Type: EmptyResponse
		Type: SetBoolRequest
			data :: bool
ERROR: LoadError: LoadError: type LineNumberNode has no field args
Stacktrace:
 [1] getproperty(::LineNumberNode, ::Symbol) at ./Base.jl:33
 [2] _addtypemember!(::Array{Expr,1}, ::String, ::String) at /home/febbo/.julia/dev/RobotOS/src/gentypes.jl:520
 [3] typecode(::String, ::Symbol, ::Array{Tuple{String,String},1}) at /home/febbo/.julia/dev/RobotOS/src/gentypes.jl:506
 [4] buildtype(::RobotOS.ROSSrvModule, ::String) at /home/febbo/.julia/dev/RobotOS/src/gentypes.jl:411
 [5] modulecode(::RobotOS.ROSSrvModule, ::Module) at /home/febbo/.julia/dev/RobotOS/src/gentypes.jl:343
 [6] buildpackage(::RobotOS.ROSPackage, ::Module) at /home/febbo/.julia/dev/RobotOS/src/gentypes.jl:302
 [7] rostypegen(::Module) at /home/febbo/.julia/dev/RobotOS/src/gentypes.jl:156
 [8] rostypegen() at /home/febbo/.julia/dev/RobotOS/src/gentypes.jl:152
 [9] top-level scope at /home/febbo/.julia/dev/RobotOS/test/typegeneration.jl:14
 [10] include(::String) at ./client.jl:457
 [11] top-level scope at /home/febbo/.julia/dev/RobotOS/test/runtests.jl:9
 [12] include(::Function, ::Module, ::String) at ./Base.jl:380
 [13] include(::Module, ::String) at ./Base.jl:368
 [14] exec_options(::Base.JLOptions) at ./client.jl:296
 [15] _start() at ./client.jl:506
in expression starting at /home/febbo/.julia/dev/RobotOS/test/typegeneration.jl:14
in expression starting at /home/febbo/.julia/dev/RobotOS/test/runtests.jl:9
febbo@febbo:~/.julia/dev/RobotOS/test$ 
```

huckl3b3rry87, 2020-08-18:
where I built PyCall
```
               _
   _       _ _(_)_     |  Documentation: https://docs.julialang.org
  (_)     | (_) (_)    |
   _ _   _| |_  __ _   |  Type "?" for help, "]?" for Pkg help.
  | | | | | | |/ _` |  |
  | | |_| | | | (_| |  |  Version 1.5.0 (2020-08-01)
 _/ |\__'_|_|_|\__'_|  |  Official https://julialang.org/ release
|__/                   |

julia> using Pkg

julia> ENV["PYTHON"]="/usr/bin/python2.7"
"/usr/bin/python2.7"

julia> pkg"build PyCall"
   Building Conda ─→ `~/.julia/packages/Conda/3rPhK/deps/build.log`
   Building PyCall → `~/.julia/packages/PyCall/zqDXB/deps/build.log`

julia> 
```

huckl3b3rry87, 2020-08-18:
where I am also running into this issue again https://github.com/jdlangs/RobotOS.jl/issues/54

huckl3b3rry87, 2020-08-18:
the expression looks like this
```
Array{Expr}((5,))
  1: Expr
    head: Symbol struct
    args: Array{Any}((3,))
      1: Bool true
      2: Expr
        head: Symbol <:
        args: Array{Any}((2,))
          1: Symbol SetBoolRequest
          2: Symbol AbstractSrv
      3: Expr
        head: Symbol block
        args: Array{Any}((1,))
          1: LineNumberNode
            line: Int64 457
            file: Symbol /home/febbo/.julia/dev/RobotOS/src/gentypes.jl
  2: Expr
    head: Symbol function
    args: Array{Any}((2,))
      1: Expr
        head: Symbol call
        args: Array{Any}((1,))
          1: Symbol SetBoolRequest
      2: Expr
        head: Symbol block
        args: Array{Any}((3,))
          1: LineNumberNode
            line: Int64 462
            file: Symbol /home/febbo/.julia/dev/RobotOS/src/gentypes.jl
          2: LineNumberNode
            line: Int64 463
            file: Symbol /home/febbo/.julia/dev/RobotOS/src/gentypes.jl
          3: Expr
            head: Symbol call
            args: Array{Any}((1,))
              1: Symbol SetBoolRequest
  3: Expr
    head: Symbol function
    args: Array{Any}((2,))
      1: Expr
        head: Symbol call
        args: Array{Any}((3,))
          1: Symbol convert
          2: Expr
            head: Symbol ::
            args: Array{Any}((1,))
              1: Expr
                head: Symbol curly
                args: Array{Any}((2,))
          3: Expr
            head: Symbol ::
            args: Array{Any}((2,))
              1: Symbol o
              2: Symbol SetBoolRequest
      2: Expr
        head: Symbol block
        args: Array{Any}((5,))
          1: LineNumberNode
            line: Int64 471
            file: Symbol /home/febbo/.julia/dev/RobotOS/src/gentypes.jl
          2: LineNumberNode
            line: Int64 472
            file: Symbol /home/febbo/.julia/dev/RobotOS/src/gentypes.jl
          3: Expr
            head: Symbol =
            args: Array{Any}((2,))
              1: Symbol py
              2: Expr
                head: Symbol call
                args: Array{Any}((3,))
          4: LineNumberNode
            line: Int64 474
            file: Symbol /home/febbo/.julia/dev/RobotOS/src/gentypes.jl
          5: Symbol py
  4: Expr
    head: Symbol function
    args: Array{Any}((2,))
      1: Expr
        head: Symbol call
        args: Array{Any}((3,))
          1: Symbol convert
          2: Expr
            head: Symbol ::
            args: Array{Any}((2,))
              1: Symbol jlt
              2: Expr
                head: Symbol curly
                args: Array{Any}((2,))
          3: Expr
            head: Symbol ::
            args: Array{Any}((2,))
              1: Symbol o
              2: Symbol PyObject
      2: Expr
        head: Symbol block
        args: Array{Any}((7,))
          1: LineNumberNode
            line: Int64 479
            file: Symbol /home/febbo/.julia/dev/RobotOS/src/gentypes.jl
          2: LineNumberNode
            line: Int64 480
            file: Symbol /home/febbo/.julia/dev/RobotOS/src/gentypes.jl
          3: Expr
            head: Symbol if
            args: Array{Any}((2,))
              1: Expr
                head: Symbol call
                args: Array{Any}((3,))
              2: Expr
                head: Symbol block
                args: Array{Any}((2,))
          4: LineNumberNode
            line: Int64 483
            file: Symbol /home/febbo/.julia/dev/RobotOS/src/gentypes.jl
          5: Expr
            head: Symbol =
            args: Array{Any}((2,))
              1: Symbol jl
              2: Expr
                head: Symbol call
                args: Array{Any}((1,))
          6: LineNumberNode
            line: Int64 485
            file: Symbol /home/febbo/.julia/dev/RobotOS/src/gentypes.jl
          7: Symbol jl
  5: Expr
    head: Symbol function
    args: Array{Any}((2,))
      1: Expr
        head: Symbol call
        args: Array{Any}((3,))
          1: Symbol getproperty
          2: Expr
            head: Symbol ::
            args: Array{Any}((1,))
              1: Expr
                head: Symbol curly
                args: Array{Any}((2,))
          3: Expr
            head: Symbol ::
            args: Array{Any}((2,))
              1: Symbol s
              2: Symbol Symbol
      2: Expr
        head: Symbol block
        args: Array{Any}((3,))
          1: LineNumberNode
            line: Int64 490
            file: Symbol /home/febbo/.julia/dev/RobotOS/src/gentypes.jl
          2: LineNumberNode
            line: Int64 491
            file: Symbol /home/febbo/.julia/dev/RobotOS/src/gentypes.jl
          3: Expr
            head: Symbol try
            args: Array{Any}((3,))
              1: Expr
                head: Symbol block
                args: Array{Any}((2,))
              2: Symbol ex
              3: Expr
                head: Symbol block
                args: Array{Any}((4,))
```
@jdlangs can you please take a look at this? this link might be helpful https://discourse.julialang.org/t/loaderror-in-macro-on-0-7/12862

huckl3b3rry87, 2020-08-18:
the error makes sense if you look at the field 
```
2: LineNumberNode
            line: Int64 463
            file: Symbol /home/febbo/.julia/dev/RobotOS/src/gentypes.jl
```
I am not sure if you are trying to get both of these, line and file and it should be an array, but now it changed

jdlangs, 2020-08-18:
I can reproduce your error, it looks like something was changed for 1.5 about the way `Expr` objects are created and some hardcoded accessing of the arguments no longer works. I've suspected that stuff was pretty brittle for a while now and I can probably easily rewrite it to not have to access the expression objects at all.

huckl3b3rry87, 2020-08-18:
That would be super helpful. I tried some last night and all morning to fix
this, but I don't know what I am looking at. And unfortunately for me I am
running into all kinds of other versioning issues with the other software
that I am trying to use which makes using Julia 1.5.0 my only choice,
unless I fix other broken things and make patches....

huckl3b3rry87, 2020-08-19:
Josh,

I know you are probably very busy, but I was wondering when you might be
able to take a look at this? And if you are not sure and it is very low
priority if you can take a look to see if we can make a patch for now, just
so that I can get my environment setup.

Thank you very much,
Huck

jdlangs, 2020-08-19:
Hi Huck,

Didn't see this message until just now but I just pushed a fixed and got a new version registered in the general registry. So you should be good once you run a package update. Let me know if you're still running into any problems.

closed 2020-08-19 by jdlangs

huckl3b3rry87, 2020-08-19:
Josh,

Wow that is awesome! Thank you so much!

I will test soon:)