jdlangs/RobotOS.jl #89: Error: Unable to load the 'rospy' python package!

JinraeKim, 2020-10-12:
Hi, I'm a newbie of ROS and think this package cool as a Julia user.

But it's little bit hard to follow the [instruction](https://jdlangs.github.io/RobotOS.jl/stable/).
So I opened this issue, and hope some detailed instruction will be added to the instruction.

## What I did
1) (at the Root env of Julia) `] add RobotOS`
2) `using RobotOS` gives me this error:
```julia
┌ Error: Unable to load the 'rospy' python package!
│ Has an environment setup script been run?
└ @ RobotOS ~/.julia/packages/RobotOS/j0Tsl/src/RobotOS.jl:35
```

## Questions
- Do I have to install rospy manually? If so, how can I install it?
- Can I use [Conda.jl](https://github.com/JuliaPy/Conda.jl)?
    - I tried to install rospy via Conda.jl but encountered this error:
```julia
Collecting package metadata (current_repodata.json): done
Solving environment: failed with initial frozen solve. Retrying with flexible solve.
Collecting package metadata (repodata.json): done
Solving environment: failed with initial frozen solve. Retrying with flexible solve.

PackagesNotFoundError: The following packages are not available from current channels:

  - rospy

Current channels:

  - https://repo.anaconda.com/pkgs/main/linux-64
  - https://repo.anaconda.com/pkgs/main/noarch
  - https://repo.anaconda.com/pkgs/r/linux-64
  - https://repo.anaconda.com/pkgs/r/noarch

To search for alternate channels that may provide the conda package you're
looking for, navigate to

    https://anaconda.org

and use the search bar at the top of the page.
```
- Can I use other Julia envs instead of Root env?

# Note
## Version
- Julia: 1.5.2
## Env
- Ubuntu 20.04

JinraeKim, 2020-10-12:
I found the solution: I needed to install rospy in advance.

For newbies like me, here's how I installed:
1) Install Conda.jl and `using Conda; Conda.add("ros-rospy", channel="conda-forge") (See [here](https://anaconda.org/conda-forge/ros-rospy) for more info)
2) Then `using RobotOS` and other stuffs, e.g., `@rosimport std_msgs.msg.Header` works well!

I think it would be betther that such basic settings are included in the instruction.

Note: I can use any Julia envs and Conda.jl as well.

closed 2020-10-12 by JinraeKim

Alvinosaur, 2021-04-26:
I fixed this issue by setting a custom python path after seeing this informative error message:

```
julia> using PyCall
[ Info: Precompiling PyCall [438e738f-606a-5dbb-bf0a-cddfbfd45ab0]

julia> pyimport("rospy")
ERROR: PyError (PyImport_ImportModule

The Python package rospy could not be imported by pyimport. Usually this means
that you did not install rospy in the Python version being used by PyCall.

PyCall is currently configured to use the Julia-specific Python distribution
installed by the Conda.jl package.  To install the rospy module, you can
use `pyimport_conda("rospy", PKG)`, where PKG is the Anaconda
package the contains the module rospy, or alternatively you can use the
Conda package directly (via `using Conda` followed by `Conda.add` etcetera).

Alternatively, if you want to use a different Python distribution on your
system, such as a system-wide Python (as opposed to the Julia-specific Python),
you can re-configure PyCall with that Python.   As explained in the PyCall
documentation, set ENV["PYTHON"] to the path/name of the python executable
you want to use, run Pkg.build("PyCall"), and re-launch Julia.
```