jdlangs/RobotOS.jl #92: RobotOS.jl doesn't work with PackageCompiler.jl

rcnlee, 2021-03-03:
I would like to speed up RobotOS startup time by compiling using PackageCompiler.  I am able to create a sysimage by passing in :RobotOS as a package to be compiled.  However, when I try to run a ROS node, the node and subscribers get started, but no messages come in.  Exactly the same code runs fine without package compilation.  

I've traced it down to ``ros_callbacks.py`` in the ``storemsg`` function.  It's getting stuck at ``self._cb_notify(self._notify_handle)`` and doesn't return.  As a result, the callbacks don't get triggered.  Any ideas?

Any help would be greatly appreciated, thanks!

jdlangs, 2021-03-03:
Unfortunately I don't have any suggestions for what to fix. The root cause probably has some connection to the Julia function that Python is trying to call being compiled in the module init, which must be a problem for the package compiler.

https://github.com/jdlangs/RobotOS.jl/blob/21a7088461a21bc9b24cd2763254d5043d5b1800/src/RobotOS.jl#L45

If you dig into this further, let me know if I can help explain anything if it's not clear what the current code is doing.

rcnlee, 2021-03-03:
I was eventually able to find a fix.  It seems ``CB_NOTIFY_PTR[]`` becomes invalid after creating the sysimage.  The pointer points to some location, but it cannot be used.  The workaround that I found is to regenerate it by calling:

``RobotOS.CB_NOTIFY_PTR[] = @cfunction(RobotOS._callback_notify, Cint, (Ptr{Cvoid},))``

I do see that is already being done in ``__init__()``, but I don't understand why that line doesn't work under sysimage precompilation.  It doesn't work even if I explicitly call ``__init__()`` on startup.  Separating that line out and calling it externally seems to work.