jdlangs/RobotOS.jl #96: Sending custom messages in Julia

ga72kud, 2022-04-28:
struct mymsg
    a::Float64
end

It would be nice if there would be custom messages to communicate with ros

I currently struggle also by sending visualization messages Marker Arrays. Is there more documentation planned?

dehann, 2022-04-29:
Hi, I have half an answer quickly at end of day.  We used custom msg a long while back.  Closest code I could find is here:
https://github.com/JuliaRobotics/Caesar.jl/blob/f10c0dc3b9e0309ebbec4ce714a8c9493e8ddba8/examples/marine/asv/rex/RExFeed.jl#L27

The `rostypegen()` does the work, but you need to link the custom types before

Here is some more documentation we've been able to capture so far (not specific to custom msg types):
https://juliarobotics.org/Caesar.jl/latest/examples/using_ros/

jdlangs, 2022-05-04:
For the visualization msgs, if you're running into problems, I would advise to try the same thing in a python script and make sure it works there. This package is quite a thin layer on top of `rospy` and shouldn't be considered an independent path to learning ROS.

ga72kud, 2022-05-05:
Thank you both. I figured out MultiArray msg work for my problem

closed 2022-05-05 by ga72kud